OpenRoberta/openroberta-lab #669: Different Robots Have Different Blocks for Beginner Mode

nishanth1232, 2020-01-22:
**Describe the bug**
 - The robots in the lab have different blocks in beginner mode. For example, the WeDo robot has the 'random integer' block in beginner mode whereas the EV3 robot does not.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch to WeDo and check available beginner blocks.
2. Switch to EV3 and check available beginner blocks.
3. See the difference.

**Expected behavior**
 - The same blocks in beginner mode.

**Screenshots**
 - EV3:
[image]

- WeDo:
[image]

**Device information**
 - Type: Laptop
 - OS: Windows 10
 - Browser: Google Chrome

KevinL10, 2020-01-24:
For other robot types the random integer block is in expert. In WeDo and mbed, it is in both expert and beginner, so removing that block from beginner would make more sense than adding it to Ev3.

boonto, 2020-02-17:
This is actually intended due to some requests from the users' side. Mbed and WeDo should have these blocks in the beginner toolbox.

closed 2020-02-17 by boonto