OpenRoberta/openroberta-lab #678: BOB3 build succeeds even with errors

Caffetaria, 2020-01-22:
**Describe the bug**
 - BOB3 build succeeds even if there is no source code.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new BOB3 program.
2. Go to the source code editor.
3. Build program
4. See error

**Expected behavior**
 -The build should fail and the error message should say: "Your program has errors so it cannot be compiled. The compiler messages are:"

**Screenshots**
 -
[image]

**Device information**
 - Type: Desktop
 - OS: MacOS
 - Browser: Chrome

boonto, 2020-01-23:
I don't think that is a bug, empty programs are valid programs as well.

Does the screenshot show what you would expect? I get the message as described in the text.

Caffetaria, 2020-01-23:
The screenshot shows the expected behaviour(as modified now).
I was comparing against the behaviour of the EV3 which throws an error if you attempt to build an empty program. 

However, when I re-checked, I realised EV3 has a main method in that particular file whose contents when deleted throw an error.
However, this is not the case with BOB3.
So, BOB3 currently shows the expected behaviour

Please let me know if this train of thought is correct and I can close the issue.

boonto, 2020-01-23:
As BOB3 is a C++ based robot the compiler accepts empty files, this is not the case for EV3 Lejos, which is Java based.
I think this shouldn't be a problem then.

closed 2020-01-23 by boonto

Caffetaria, 2020-01-23:
Ah, sorry, I was referring to EV3 c4ev3.
But yes, EV3 c43ev3 has a main method in the program whereas BOB3 does not, so I see why this isn't a problem.
Thanks :)

boonto, 2020-01-23:
In the BOB3 library files given to the compiler there is a file which specifies the main method, this is not the case for c4ev3. That's why it is necessary for the c4ev3 :)

Caffetaria, 2020-01-23:
Oh yes, thanks :)