OpenRoberta/openroberta-lab #680: Addition of Continuous Images

nishanth1232, 2020-01-22:
**Is your feature request related to a problem? Please describe.**
 - When the 'shift image' block is used to shift an image out of the robot's display, the part that is shifted out disappears.

**Describe the solution you'd like**
 - A block that shift's the image in a continuous manner, that is, when the image is shifted down, the part that disappears should reappear at the top. Similarly, it would be nice to have the feature for up, left and right as well.

**Describe alternatives you've considered**
 - Addition of an option in the 'shift image' block that enables continuous images.

RohanCheri, 2020-01-23:
I have been able to add in the block to the blockly blocks file and have added it to the interpreter (As shown below).
[image]

I have also been able to add the block to the MBed validator, generator and worker files but I am unsure of what to add in the specific visitor files to compile into another language. Concerning the Micro: Bit python visitor, there is no function currently in the API to achieve this task (I believe it could be done through the use of the basic shift functions and the crop function). Additionally, with the Calliope API, it seems that this block would be very similar to the scroll function (Which actually seems to be a loop so it might require a different implementation).

boonto, 2020-01-24:
I think this is a lot harder than one would initially suspect. 
You would have to save the image in a temporary array and read the shifted out pixels from the array.
But the real difficulty would be to properly implement all the different interactions of multiple shift / invert blocks.

RohanCheri, 2020-01-24:
[image]
This is currently how the code looks. The reason that this would work is that all the information that is poped or shifted out is directly added back into the opposite side of the list (pop and shift both return the value(s) that have been removed, meaning that there is no need for a temporary array as the addition can just be done directly).
The rest of the code is the same as in the normal shift image action which allows for the code to also handle multiple shifts. Additionally, this also seems as if it would work with Invert and other image functions as it follows almost the exact same function structure as the normal shift image action with only a few minor changes to how the shifting occurs meaning that it should theoretically act as almost the same as the normal shift action, which currently works with all of the image functions available.

boonto, 2020-01-24:
Ah, I misunderstood, that is way easier, at least for Javascript.
For the C robots it would be a little bit harder I assume, but not impossible.

RohanCheri, 2020-01-24:
I will look into how to do this in C. 
Just a question, where would the code for these extra functions be stored so that the robot can use them? 
I would assume somewhere in the visitor file or the NEPOdefs.sh (on a side note, where is the information for this file stored?)

boonto, 2020-01-24:
The NEPOdefs.h for Calliope 2017 can be found [here](https://github.com/OpenRoberta/ora-cc-rsc/blob/master/RobotMbed/libs2017/microbit/inc/NEPODefs.h) in the ora-cc-rsc repo.

RohanCheri, 2020-01-24:
So would i push to the ora-xc-rsc repo to create this new function in C?

boonto, 2020-01-24:
Yes, that is specified in the run configuration and used by the cross compilers to build the program.

bjost2s, 2020-01-24:
Unfortunately mbed is holding always one image to display, so once you shift it, you do not get a new, shifted image but you changed the one in place. Please have a look how this is solved for mbot, where you always keep the origin one as it is and get a new image object after shifting etc. 
So changes might be in the microbit dal where the MicroBitImage class is defined.

RohanCheri, 2020-01-24:
@bjost2s Oh okay that makes sense. Just a clarification, are you talking about storing an image in the front-end or back-end of the code? (One more thing, I see that you said that the changes would be stored in the "microbit dal..." where would that be located?)

RohanCheri, 2020-01-24:
Also, while working on this project, I believe that it actually might be more appropriate to just add another option to the existing image function rather than creating a whole new block.
Would this be fine?

RohanCheri, 2020-01-25:
@bjost2s @boonto So I have decided on making this an option that comes from a drop-down menu attached to the block. I have gone ahead and added this new option for the block in all the necessary places (other than the code generator I still need to work on making the shifting functions in C). I have attached a video of it running below.
[image]

How does this look so far?

NOTE: The shift works for all directions :)

RohanCheri, 2020-01-25:
One more thing... in order to make the Blockly blocks easier to edit, I had gone ahead and uncompressed the file. So How would I go about re-compressing the blockly files?

boonto, 2020-01-27:
Looking good! Another problem is running it on the robot itself, as @bjost2s mentioned the API has its own handling of the image storage which might not work out of the box.

Blockly development usually happens in our fork of the [blockly](https://github.com/OpenRoberta/blockly) repository. There you need to run the python build script. However, there is a dependency to the clojure library and we always had problems building it on Windows machines.

RohanCheri, 2020-01-27:
@boonto @bjost2s Would it be appropriate to add a new parameter in all of the shift function which indicates whether or not the shift is continuous or not. (This would also require a symmetric change in the Visitor files in order to properly compile the code with the new parameter). Additionally, where is that actual implementation code of the function, I am able to see the function stems such as "int shiftDown" but I am not able to see the actual code that is used within it.

boonto, 2020-01-29:
There should be specific implementations in the micro:bit API and the Calliope API.