OpenRoberta/openroberta-lab #685: Addition of 'Ctrl+S' Save Feature

nishanth1232, 2020-01-23:
**Is your feature request related to a problem? Please describe.**
 - In the lab, when the user is logged in and Ctrl+S is clicked, the program does not save into 'my programs' and instead the HTML extension of the web-page is saved to the computer.

**Describe the solution you'd like**
 - Addition of 'Save' feature when Ctrl+S is clicked and user is logged in.

**Additional context**
 - It can open the 'save as' dialog box when Ctrl+S is clicked and the user is logged in. If the user is not logged in however, it could save into the user's computer.

ani1998ket, 2020-02-23:
Hi, 
I would like to take up this issue.
I already solved it on developer console of the website.
Could you please guide me to solve the issue.

boonto, 2020-02-26:
Do you have a local clone of the repo yet? Make the changes and test them there and then you can create a pull request.

ani1998ket, 2020-02-26:
yes i have a clone. I found out that the js files are located in OpenRobertaServer / staticResources / js folder. Just wanted to know where should i add the event listener function. Also is there a function to save the project such that when Ctrl + S event is triggered i can call it thus it will save the project instead of the html file

boonto, 2020-02-26:
You can do it in the same way as the other hotkeys, look up how they work and try to do it similarly. Use the "Find in all files" function of your IDE to locate it.
In order to properly save you need to trigger the appropriate REST call. You should be able to find it and how they work by looking through the files.
If you need any more pointers feel free to ask.

boonto, 2020-09-02:
We have decided to slightly restructure the scope of this issue. Instead of only focusing on the CTRL + S save feature, there should be a more general concept and implementation for keyboard shortcuts.
Questions:
What features should be accessible through keyboard shortcuts? (Import/Export)
What shortcuts would best represent these actions?
How to show the shortcut to the user?
How to handle devices without immediately accessible keyboards?

These questions (and maybe other ideas?) should be thought through before the implementation. Once properly specified and discussed the feature can be implemented.

This should still fall in the scope of "good first issue" as it gives some insights into the Lab and how users might use it, while providing a fairly straightforward frontend implementation effort, in a kind of mini project.

closed 2020-12-21 by bjost2s