OpenRoberta/openroberta-lab #728: Bookmarks

RichardBotez, 2020-01-30:
- The bookmark selection box found at the bottom of the NEPO tab does not appear to complete its purpose and the bookmark option cannot be found afterwards.

** Describe the solution you would like **
- A more concise description of this function will be handy.

**Additional context**
[image]
[image]

boonto, 2020-02-04:
At that point you are supposed to create a bookmark using your browser.
As this cannot be done automatically, we can only do it that way.

Is the message unclear?

RichardBotez, 2020-02-04:
The bookmark to which it refers is unclear. I think adding "chosen browser bookmark" to the message will help.

closed 2020-08-20 by rbudde