OpenRoberta/openroberta-lab #730: Implement update workflow for ev3dev

ensonic, 2020-01-31:
Users of the ev3dev system need to reflash their sdcards to get new features. They also needs to figure by themself that an update is required.

In order to improve this we should do the following:

- [ ] implement server side logic to check the version of the connecting robot against expected version. Initially this could just log the version and if an update should be triggered. TODO: add link to server side code that needs to be changed.
- [ ] implement a notification form the web frontend. The implementation will tell that a new version is available and will inform for how long the currently used version will be supported. The notification will link to a webpage with install/upgrade instructions.
- [ ]  implement the connector download feature where an existing firmware can be updated on-the-fly. For this the steps outlined in [robertalab-ev3dev](https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/lab.py#L412_L419) need to be implemented
- [ ] document the process of releasing an connector update (lejeos jar or ev3dev python)
- [ ]  release a new ev3dev image with the above features and the enable the notification on the frontend, asking people to reflash

boonto, 2020-02-07:
Frontend wise it is handled in the [handleFirmwareConflict](https://github.com/OpenRoberta/openroberta-lab/blob/master/OpenRobertaServer/staticResources/js/app/roberta/controller/robot.controller.js#L277) in `robot.controller.js`. Version comparison is done in [ClientAdmin](https://github.com/OpenRoberta/openroberta-lab/blob/master/OpenRobertaServer/src/main/java/de/fhg/iais/roberta/javaServer/restServices/all/controller/ClientAdmin.java#L202).

The "protocol" is pretty weird right now as the robot requests the exact files through one REST call each.
I think it would be nice to revise it while handling this (for now only ev3dev).
The server could tell the robot which files are necessary, instead. This would decouple the actual robot files from the protocol and the robot firmware.

ensonic, 2020-02-07:
I was actually considering to alway fetch a zip with the python files (2 py files with code and one with the version info). This way the protocol would stay simple - just one file (zip) and the set of files downloaded is atomic.

boonto, 2020-02-07:
Sounds nice as well, that way its very simple and variable files are allowed.

ensonic, 2020-02-07:
I've got the server side parts working:
https://gist.github.com/ensonic/b19ee4229185f8923a0e6001fdc786bd

Also on the ev3dev the download is received and updated. I need to do some code cleanups, the restarts of the connector and then making a release.

ensonic, 2020-02-10:
PR ready. This can be safely merged (won't do anything yet). What we need to discuss is how to add an extra check so that if the ev3dev menuVersion is <1.7.5 we prompt people to reflash the image (with url to image) and only if menuVersion is >=1.7.5 we do the normal updates.