OpenRoberta/openroberta-lab #738: Code function block

philippmaurer, 2020-02-10:
**Is your feature request related to a problem? Please describe.**
There are some instances, in which it is not possible to solve all program tasks with the given NEPO blocks, especially when it is about custom sensors/actuators for Arduino based systems.

**Describe the solution you'd like**
I would like to suggest a block, with which you can create a function in your NEPO program, but define its code with textual code of the target system. (C/C++, Java, Python)
This would give teachers or hardware developers the opportunity to create solutions for small problems that can not be solved with NEPO code and make its solution accessible for their students.

[image]

Being able to only provide a function evades possible problems with Python indenting and possible conflicts with other globally defined variables for all other languages. Concerning function parameters the NEPO syntax can be used.

**Describe alternatives you've considered**
Creating a ticket here and waiting until a specific sensor/actuator was developed by the community, or solve the ticket yourself. This is not something teachers are most likely willing to do and it is not suitable for all instances, since the amout of available blocks need to be clear.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

philippmaurer, 2020-02-10:
Possibly a duplicate to #121, but i could not get access to the linked google doc, so maybe it is, maybe not.

rbudde, 2020-02-10:
no duplicate.

The old issue asks for fragments, that can be stored (currently only complete proprams can be stored) and "copied" into a program.

The new issue allows native code to be written for functions and is one way to add new functionalitiy (e.g. bit ops or in a limited way new sensors/actors) to the lab