OpenRoberta/openroberta-lab #741: ev3 sim move block does not work

rbudde, 2020-02-12:
write an ev3 program:

- move 30%
- wait 5000 msec

This did work in 3.8.1, but doesn't move in 3.8.2

Heep042, 2020-02-12:
This is related to https://github.com/OpenRoberta/openroberta-lab/commit/7b1a967acdf811999165e0c6b106e8e9ad1d098c reverting the commit solves the issue. It appears that https://github.com/OpenRoberta/openroberta-lab/issues/492 needs further changes to be fixed.

boonto, 2020-02-12:
Thanks, I found that as well.
Previously, the blocks without distance just added distance `0` as a dummy, which lead to this issue in combination with 7b1a967.
I improved the previous implementation so that the distance is only pushed to the stack when it is necessary.

rbudde, 2020-02-12:
tested for both simulation and WeDo (which share this implementation). A property added twice should be removed later (only a cosmetic repair)

closed 2020-02-12 by rbudde