OpenRoberta/openroberta-lab #748: Old Logo usage in "about Open Roberta Lab" help popup

philippmaurer, 2020-02-17:
**Describe the bug**
The "about the Open Roberta Lab" popup still shows the Open Roberta logo.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the ORLab
2. Open the help menu
3. Click on "about the Open Roberta Lab"
4. See the old logo

**Expected behavior**
The "new" logo should be shown.

**Screenshots**
[image]

bjost2s, 2020-02-18:
The open roberta lab logo is the actual one! Please undo!

bjost2s, 2020-04-21:
new logo:
[image]

closed 2020-04-22 by bjost2s