OpenRoberta/openroberta-lab #750: mBot: connecting via Bluetooth

Theaterfuchs, 2020-02-19:
**Is your feature request related to a problem? Please describe.**
It needs some steps to connect the pc and the mBot via USB using tokens etc.

**Describe the solution you'd like**
I would like to transfer the programs from open.roberta to the mBot via bluetooth

**Describe alternatives you've considered**
I looked for a radio-connecting USB-A connection. In effekt I would like to let the mBot on the floor while i am programming on the desk.

**Additional context**
 I have chosen the mBot for artificial reasons. He should appear on stage but look like a robot and not like a toy. Furthermore I want to deploy a lot of sensors wich the lego robots do not have notwithstanding the lego robots do connect better to the roberta eco system

bjost2s, 2020-02-23:
mBot works like an Arduino Uno, that means the user programs are transferred completely to the robot. Currently it is not possible to do this via Bluetooth because of the limitations of the bootloader/firmware. AFAIK there is no solution, do you have an idea how this may work?

Theaterfuchs, 2020-02-23:
[image]

I had a look to wireless **USB A to USB B** connection. 
Finding such a thing I would attach the recieving unit on the mBot.
"Bin am suchen am dran am sein…" [still searching…]

Theaterfuchs, 2020-02-23:
[image]

Combining this with the thing above. Maybe I will achieve it and test it.