OpenRoberta/openroberta-lab #752: Gallery improvements

rbudde, 2020-02-26:
the number of programs in the gallery increases. The overview about all the entries is more and more difficult. Possible improvements:

- restrict the programs shown to the robot actually selected (this may be a filter, that can be changed by the user. I'd prefer a hard restriction: why should a Calliope user see Ev3 programs? If he/she wishes to see them, a second tab with another robot is enough)
- allow switching from tiles to lines and vice versa
- add "column" filter (as already supported by bootstrap tables)
- add sorting (as already supported by bootstrap tables)
- it may make sense to reduce the size of the tiles: robot image together with the name, views/likes in one line or as popup
-the user should be able to select the tile color

Issues #349 and #719 and #564 and #690 are related to this issue

abhimnc, 2020-02-29:
hi, I am a newbie here and want to contribute. This is mentioned as a good first issue. Can anyone guide me on how to proceed further to resolve? Where to look for in the code base?Thanks!

boonto, 2020-03-02:
Take a look at Bootstrap tables in general, then you can hack around in the `galleryList.controller.js` file to get a feel for the capabilities of the tables.

ensonic, 2020-03-03:
+1 for having a default filter to the selected robot applied.

ykabusalah, 2020-03-23:
can i get a shot at this?

boonto, 2020-03-24:
Sure, what about #557? Should I remove your assignment there?

rbudde, 2020-08-20:
some improvements have to be done, thus NOT closed.
Nevertheless the gallery is much better now.

rbudde, 2020-08-26:
Moved to be done improvement to a new issue. Thus can be closed.

closed 2020-08-26 by rbudde