OpenRoberta/openroberta-lab #763: EV3 test block generates incorrect source code

Heep042, 2020-03-04:
**Describe the bug**
The test block is missing parenthesis, thus, when used inside arithmetic operations, incorrect result gets produced

**To Reproduce**
Print out such expression:
test(true, 0, 1) - 1

see that the java code produces something in the lines of:
true ? 0 : 1 - 1

**Expected behavior**
Result of -1, and properly parenthesised java code ((true ? 0 : 1) - 1)

iamyashm, 2020-03-09:
Hi. I would love to work on this. Could you please assign this to me and, if possible, guide me regarding where I should start exploring the codebase from?

boonto, 2020-03-09:
Hey, take a look at the code generation visitor for EV3 and the more generic Java visitor. You should be able to find it easily.

iamyashm, 2020-03-09:
Could @Heep042 please give a test case where the output is not as expected? Because even though the source code generated does not have parentheses, the result of the expression is still as expected.

iamyashm, 2020-03-09:
Would parenthesising the test block unconditionally be a good solution for this?

Heep042, 2020-03-09:
It would be the simplest solution with no side-effects and potential bugs in the checks. I would say yes.

iamyashm, 2020-03-09:
I've changed the code to add parentheses but as a result, a test is failing on build since it expects only 1 pair of parentheses. How do I fix this? Do I need to alter the test itself or is there a way to bypass it so that I can run my code?

boonto, 2020-03-09:
Yes, you should change the test as well as now the parenthesis are expected to be there all the time.

iamyashm, 2020-03-09:
Thank you for your help! I've made the necessary changes and submitted a PR.

rbudde, 2020-03-17:
works as expected

PR #768 (2020-03-10): Test block paranthesis
fixes #763

I have updated the Java Visitor to always enclose a test block (i.e. ternary expression) inside parentheses. Additionally, the Logical Expression test has also been modified to include parentheses around the ternary expression.
This is my first contribution, please let me know if you have any feedback!
closed 2020-03-18 by rbudde