OpenRoberta/openroberta-lab #764: Backtransformation of ConfigurationComponent

boonto, 2020-03-05:
In #182 `ConfigurationComponent` was modified to contain blockly information. However, the backtransformation generates invalid Blockly XML.
This should be modified to generate valid XML for all configuration blocks, so that the annotations can be used for future configuration checkers.

- [ ] Add round trip tests to validate transformation and backtransformation of all configuration blocks
- [ ] Implement backtransformation as generic as possible

boonto, 2020-03-10:
The configuration components of the new configuration currently have no distinction in the XML whether they are actors or sensors.
Currently there is no distinction needed, as all robots with the new configuration support actors and sensors on all available ports (Arduino, Festo, Sensebox, WeDo, soon Calliope).
This should be necessary once serverside configuration validators want to distinguish between actor/sensor ports e.g. on EV3. Here, I believe a non XML solution would be preferred, e.g. in the YAML.

bjost2s, 2020-05-07:
see #806 works fine now!

closed 2020-05-07 by bjost2s