OpenRoberta/openroberta-lab #784: Help files improvements

bjost2s, 2020-03-26:
The help files (html) are derived from our [Open Roberta Wiki](https://wiki.open-roberta.org), Here we only have two sites (one in German, one in English) which describes all common blocks. A common block is each block that does not appear in either the Action, Sensor or Message categories.
Some supported Systems, like e.g. WeDo or Bob3, have not implemented all of these common blocks. For this we will add a special CSS class, starting with "not" and followed by the groups name of the systems, e.g. "notWedo", and set them to be invisible.

phoenix-kd, 2020-03-30:
i am unable to locate these help files. Can you help me with their exact location?

sachinsshetty, 2020-03-31:
> i am unable to locate these help files. Can you help me with their exact location?
@phoenix-kd 

Changes are made in this commit. [link](https://github.com/OpenRoberta/openroberta-lab/commit/f977f387ce3d6b10a1850f871f59a07e91f15940) 
Files are 
* OpenRobertaServer/staticResources/help/progHelp_bob3_de.html 
* OpenRobertaServer/staticResources/help/progHelp_wedo_de.html
* OpenRobertaServer/staticResources/help/progHelp_wedo_en.html 
* OpenRobertaServer/staticResources/js/app/roberta/controller/progHelp.controller.js

phoenix-kd, 2020-03-31:
Made some changes to the files [here](https://github.com/OpenRoberta/openroberta-lab/pull/788/commits/1a9c7c36dbf2b62ce659bd564d8f685188f55cb2)
Open to suggestions
PS: Had to submit an early incomplete PR as I am applying in #GSoC2020 and took this as a first issue to work on!

boonto, 2020-03-31:
The help files themselves are autogenerated from the wiki, which is maintained by one of our colleagues.
As for the aspects stated in the issue, @bjost2s is working on those.

For general proposed changes to the help files (and thus the wiki) like typos or grammatical errors you can send us any issues you find and we will pass it on to our colleague to incorporate into the wiki.

boonto, 2020-04-02:
Help files for BOB3 and Wedo show up now.

closed 2020-04-02 by boonto