OpenRoberta/openroberta-lab #785: As a teacher I would like to create and manage groups for my classes to be better prepared in the lessons

bjost2s, 2020-03-30:


bjost2s, 2020-07-09:
@philippmaurer , currently there is an issue when I delete the user (teacher account) while one of the students is still logged in

1. The teacher can delete his account although he holds accounts for groups with active members
2. The student (while logged in) gets 
[image]
instead of a dedicated message about his account which does not exist anymore.

rbudde, 2020-07-10:
Sorry for the german text ... its C&P from a discussion:
ich finde es SEHR irritierend, dass mein Gruppenaccountname z.B. g:03 ist (siehe Statusinformation) ist, ich aber 03 beim login eingeben muss (und nicht g:03), mein Passwort aber g:03 ist. Der Lehrer sieht ebenfalls in seiner Liste g:03 und wird dem Schüler bestimmt einen falschen Tipp geben, wenn der seine Accountnummer vergessen hat. Mein Vorschlag: im Gruppenlogin 'Name der Gruppe' WEGLASSEN und als Account g:03 eingeben (z.B. Klasse9:Reinhard).

closed 2020-08-20 by rbudde