OpenRoberta/openroberta-lab #786: Variable names should be defined properly

rishabhdeepsingh, 2020-03-30:
**Describe the bug**
 - 

**To Reproduce**
Steps to reproduce the behavior:
1. Click on the <kbd>+</kbd> begore the start block on the main page.
2. type "  for(int i = 0; i < 10; ++i)" without quotes.
3. Press <kbd>Enter</kbd>

**Expected behavior**
 - Error or warning showing that invalid variable name.

**Screenshots**
[image]
[image]

**Device information**
 - Type Desktop
 - OS: Catalina 10.15.3
 - Browser chrome Version 80.0.3987.149 (Official Build) (64-bit)

**Additional context**
 - The possible workout would be to add a RegEx checker for the variable name, and if variable name is not following RegEx a warning should be shown.

bjost2s, 2020-03-31:
There are checkers, both on frontend and backend site. I cannot reproduce this behaviour. Will check it on browser stack

bjost2s, 2020-03-31:
Could you please make a screenshot after clicking somewhere else than in the variables name field and put it in here?

rishabhdeepsingh, 2020-03-31:
The error comes when you click anywhere on the. 
PS: I have updated that above.
[image]

bjost2s, 2020-04-01:
which robot have you chosen?

rishabhdeepsingh, 2020-04-02:
I have selected the EV3.

bjost2s, 2020-04-03:
Thank you very much, now I can reproduce it! Seems that validators in blockly are not triggered, but validators on backend side should catch this

rishabhdeepsingh, 2020-04-12:
@bjost2s  Can you refer to some files where I can resolve this issue? 
I tried finding some files but could not find the direct link to the variable name checker.

bjost2s, 2020-04-14:
@rishabhdeepsingh, it seems that we do not have a variable name tester at the backend yet. It should be implemented as part of the empty checker, see #306.

rishabhdeepsingh, 2020-04-14:
Let me know If I can contribute on this one and where.

bjost2s, 2023-04-11:
works

closed 2023-04-11 by bjost2s