OpenRoberta/openroberta-lab #800: Gallery View

Thorlogy, 2020-04-15:
**Describe the bug**
 - It seems as if the representation of the tiles in the gallery is different from user to user. With me the representation is faulty (100% representation in the browser) with others the representation is correct. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Open Roberta Lab
2. Click on Gallery
4. See error

**Expected behavior**
 - A uniform representation of the gallery tiles on all screens

**Screenshots**
[image]

**Device information**
 - Typ: mac 
 - OS: mac
 - Browser chrome, safari,  - 

**Additional context**
 - WITH FIREFOX it works WELL!!!

bjost2s, 2020-04-18:
The fix should also resolve the problem of English labels instead of translated labels and the following strange behaviour (smartphones only)

[image]

boonto, 2020-04-20:
Works well with different screen widths. Tested with Safari and Chrome on Windows and Edge, IE, Firefox on Windows.

closed 2020-04-20 by boonto