OpenRoberta/openroberta-lab #806: Show annotated configuration in frontend

boonto, 2020-04-17:
With with #114 and #764 the backtransformed annotated configuration (currently only for Arduino) could be used in the frontend.

Here https://github.com/OpenRoberta/openroberta-lab/issues/114#issuecomment-516904543 we already described how this should look:
- [ ] rename `data` field to `program_blockset`
- [ ] rename `configuration` field to `configuration_blockset`
- [ ] show recieved configuration in workspace
- [ ] rename workspaces to better reflect their content?
- [ ] add a visual cue to the configuration tab (flashing and/or icon)

boonto, 2020-04-27:
Currently, the configuration checker for Arduino (the only one currently annotating the configuration) only checks for overlapping pins for some specific sensors/actors, that was extended to all available in the configuration.
However, this also brought up that sometimes overlapping pins are valid and might be what the users wants (e.g. 2 LEDs on one output). We discussed what to do here and decided that overlapping pins should not be allowed. This is more user-friendly for inexperienced users and experienced users may still use overlapping pins, but without having it represented in the configuration.
This may change depending on user feedback, but seems like the best decision for now.

bjost2s, 2020-05-07:
Detailed and korrect error messages are available now in the annotated configuration blocks !

closed 2020-05-07 by bjost2s