OpenRoberta/openroberta-lab #807: Arduino Stepper not working

bjost2s, 2020-04-19:
**Describe the bug**
When configuring and programming a stepper motor, the motor does not work.

**To Reproduce**
Create the following test program
[image]

**Expected behavior**
The motor should rotate

**Additional context**
 Use a correct configuration!

The program works as expected if you exchange IN2 with IN3!

rbudde, 2020-04-20:
this program failed when compiled, or not? The integration tests have to be extended?

bjost2s, 2020-04-20:
No problem with compilation, only at runtime the motor does not rotate

bjost2s, 2020-04-22:
Works with my stepper motor from the funduino set

closed 2020-04-22 by bjost2s