OpenRoberta/openroberta-lab #816: Information box overlays text input fields

Thorlogy, 2020-04-24:
**Is your feature request related to a problem? Please describe.**
I think the screenshot describes it best. 

[image]

**Describe the solution you'd like**
The info box should be wider, then the info text could go over two lines. This way you can see the other text boxes and then click on them. 

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

bjost2s, 2020-05-07:
Still a problem for small devices, because the hints are covered by the keyboard:
[image]

rbudde, 2020-05-07:
for large devices fine, for small devices much better as the solution before. I close the issue

philippmaurer, 2020-05-07:
If there is an error message in the following form field, the error message looks like it is a part of the hint for the current form field.

[image]

boonto, 2020-05-07:
I think all add the same behaviour to the `form-invalid` class. However, it is used in more places than the hints and has to be checked there as well.

boonto, 2020-05-08:
Now the error messages should not overlap anymore.

rbudde, 2020-05-11:
after entering the password again, you have to click the button `Jetzt registrieren` _twice_ to register. Not user-friendly.

boonto, 2020-05-12:
You should only need to press the button once now. A hint will only disappear when opening another hint.

closed 2020-05-12 by bjost2s