OpenRoberta/openroberta-lab #817: Reset Password: input fields not accessible when using firefox

dr0i, 2020-04-26:
On ubuntu `firefox 75` :
Using the link provided in the email after resetting password I cannot input anything in the password fields.

Other browsers (e.g. `Chromium`) work, though.

bjost2s, 2020-04-26:
We already know about this. Could you please check this https://support.mozilla.org/de/questions/1264837 and compare the last entry with your settings?

dr0i, 2020-04-27:
Thx @bjost2s for the quick answer.
Point is, when doing normal login everything is good (characters shown wehn typing username, stars when typing password). Only when doing trying to login after resetting the password nothing is shown. So the question is:
why is there a different behaviour?