OpenRoberta/openroberta-lab #828: Improvement of providing the status text

bjost2s, 2020-05-09:
The status text for the Lab is not stored permanently, e.g. in the database, yet. Each time the server is restarted, the message is not available anymore.
In best case the status text and the expiring date is stored in the database, so that it works more reliable.

boonto, 2020-09-10:
As a preliminary measure to improve communication of breaking changes for the last release I added a simple "release info" system. Currently the messages are hardcoded and not time limited, which should be improved for the full resolution of this issue.

bjost2s, 2020-09-29:
This solution is not the final solution, so it will not be closed but forwarded to the next release

closed 2020-11-02 by bjost2s