OpenRoberta/openroberta-lab #846: Docker Setup for Raspberry Pi

kraeml, 2020-06-08:
What I have done and my setup for Pi4:

```bash
git --version
```

    git version 2.20.1

```bash
java -version
```

    openjdk version "11.0.7" 2020-04-14
    OpenJDK Runtime Environment (build 11.0.7+10-post-Raspbian-3deb10u1)
    OpenJDK Server VM (build 11.0.7+10-post-Raspbian-3deb10u1, mixed mode)

```bash
cat /etc/os-release
```

    PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
    NAME="Raspbian GNU/Linux"
    VERSION_ID="10"
    VERSION="10 (buster)"
    VERSION_CODENAME=buster
    ID=raspbian
    ID_LIKE=debian
    HOME_URL="http://www.raspbian.org/"
    SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
    BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```bash
uname -a
```

    Linux pi4-home 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

```bash
free
```

                  total        used        free      shared  buff/cache   available
    Mem:        3999744      389120     2814556        8996      796068     3460564
    Swap:       1048572           0     1048572

```bash
export ARCH=arm32v7
```

```bash
echo $ARCH
```

    arm32v7

```bash
id
```

    uid=1000(pirate) gid=1000(pirate) Gruppen=1000(pirate),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),105(input),109(netdev),996(docker),997(gpio),998(i2c),999(spi)

```bash
pwd
```

    /home/pirate/notebooks/openroberta-lab

```bash
nmap localhost
```
[... 246 lines cut ...]
```bash
docker network ls
```

    NETWORK ID          NAME                DRIVER              SCOPE
    62abbc60c016        bridge              bridge              local
    ec3caaa57eb5        host                host                local
    d5d27fda432a        none                null                local
    e0ef91bd1e3f        ora-net             bridge              local

```bash
./ora.sh -dbParentdir pi-ora/db/test create-empty-db
```

    creating an empty db using the url jdbc:hsqldb:file:pi-ora/db/test/openroberta-db - an existing db is NOT overwritten
    Fehler: Hauptklasse de.fhg.iais.roberta.main.Administration konnte nicht gefunden oder geladen werden
    Ursache: java.lang.ClassNotFoundException: de.fhg.iais.roberta.main.Administration

```bash

```

I'm not able do install the db on my Pi4. Any suggestions will help.

THX Kraeml

rbudde, 2020-06-09:
thx for the detailed information. I'll add comments here to simplify our docker setup resp to help with your setup.

rbudde, 2020-06-09:
you cannot create the empty database, because the libraries needed are not yet installed. You have to run 'mvn clean install -DskipTests' (rm the -D to see the unit tests executed). Then it should work.

As a consequence I'll make the following changes:
1. improve the first paragraph of the readme
2. add an empty database to allow a simple start w.o. generating a database
3. remove the INAME parameter to have simpler names in dockerhub. Consequence: only one installation on a machine. That seems ok. This will change the image names. The database image and the server images MUST be generated again. Do not forget to remove old images, they use up a lot of space!

I push the changes to the branch "docker-setup". Feel free to evaluate it.

kraeml, 2020-06-10:
THX for your help. The pi docker setup is now working. The output of my [jupyter notebook is here](https://gist.github.com/kraeml/adc59bdc49c02ce43c0d2d7d11e9199f). I forgot the `mvn clean install -DskipTests`.

bjost2s, 2024-10-23:
solved

closed 2024-10-23 by bjost2s