OpenRoberta/openroberta-lab #847: Arduino: Add the "create text with" block

bjost2s, 2020-06-12:
Currently this block is not implemented for Arduinos nor for Arduino like plugins. If one would like to collect sensor data this block is necessary to label and structure the measurement data e.g. for the "write serial" function.
[image]
It is possible to use the "append text" block, but it makes the program very complicated and the workaround is not easy to understand.

viveksahu26, 2020-08-08:
air, i would love to work on this issue.. so please assign me..

boonto, 2020-08-10:
Hey, you can work on it without having it assigned. Did you get the project up and running yet? We have some documentation in the README.md and wiki.

bjost2s, 2023-05-08:
Tested