OpenRoberta/openroberta-lab #854: direct link to the tutorials

rbudde, 2020-07-01:
we have a direct link to the gallery. The same way we should have a direct link to the tutorials

hariom1625, 2020-07-03:
I would like to work on this, can you be more specific on where to add the link to tutorials.

rbudde, 2020-07-08:
sorry for answering so late. If you start the lab with `https://urlOfLab#linkInfo`  the `link-info` is processed in the Javascript file `menu.controller.js` . When I wrote this answer for you, I had a look into the code and found around line 45 the handling of our gallery-link, that has only to be C&P and modified for the tutorial. It is a very small change. It reads simply:
```
     } else if (target[0] === "#tutorial") {
            GUISTATE_C.setStartWithoutPopup();
            $('#tabTutorialList').click();
```
I'll add that to our codebase and mention your effort, of course. Thank you for your interest!!!

hariom1625, 2020-07-08:
No worries it is fine, that's generous of you.
I will look for other issues to solve for my contributions to this community.
Thanks for the reply though :)

rbudde, 2020-07-10:
direct link to tutorial now available, thanks to Hari Om Yadav!

iagolirapasssos, 2020-07-15:
Hello, I want to time the virtual robot time of departure and arrival, anywhere. How can I access the real time of execution of this process? I have not found any block that does this. I tried to make counters, but these do not respond linearly to real time.

boonto, 2020-07-16:
If you are talking about EV3 simulation, check out the "get value ms timer" block.

iagolirapasssos, 2020-07-20:
> If you are talking about EV3 simulation, check out the "get value ms timer" block.

Thank you! I got what I wanted with your information.

iagolirapasssos, 2020-07-20:
Is there a way to capture the coordinate values? It would be useful to create arenas with Maze Solution. Great!

boonto, 2020-07-21:
Existing issues are not really suited for these kind of questions, they are better of in our [Open Roberta Google Group](https://groups.google.com/g/open-roberta).
There is no way to get these values, you would have to implement a clever solution yourself, e.g. with a colour coded custom background and using the colour sensor to read these values.

iagolirapasssos, 2020-07-22:
I'm sorry for insisting, but I already sent a request to join the group and they never added me. It's been a long time since I sent the invitation.

boonto, 2020-07-23:
There was an issue with the notifications, that should be fixed.

closed 2020-08-20 by rbudde