OpenRoberta/openroberta-lab #858: Problem creating account

Thorlogy, 2020-07-18:
**Describe the bug**
 - I m not able to create a account at Open Roberta Lab

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 - After the registration process the browser makes a refesh, but no account has been created

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]: Mac
 - OS: [e.g. windows], including version: Mac Mojave 10.14.6
 - Browser [e.g. chrome, firefox], including version: Chrome 83.0.4103.116 (Offizieller Build) (64-Bit)

**Additional context**
 - Same procedure works fine with firefox

bjost2s, 2020-07-20:
Can neither be reproduced on Linux / Chrome nor on https://www.browserstack.com/ with MaxOSX Mojave and Chrome 83
@boonto , could you please try it on your Macbook?

Thorlogy, 2020-07-20:
solved.

Thorlogy, 2020-07-20:
solved

closed 2020-07-20 by Thorlogy