OpenRoberta/openroberta-lab #859: Menue not clickable

Thorlogy, 2020-07-18:
**Describe the bug**
 - A clear and concise description of what the bug is.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 - A clear and concise description of what you expected to happen.

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]: Mac
 - OS: [e.g. windows], including version: Mojave 10.16.4
 - Browser [e.g. chrome, firefox], including version: Firefox 

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2020-07-20:
strange, because it should look like:
[image]
from browserstack.com MacOSX Mojave Chrome 83

RaghuvirShirodkar, 2022-10-04:
Seems to be resolved. There is now an option to mention number of multiple sims.

Tested on BrowserStack (MacOS Mojave):
[image]

bjost2s, 2022-10-04:
@Thorlogy is the problem solved? Please check it again. If there is still an issue, please document it in reproducable way.

Thorlogy, 2022-10-04:
problem seems to be solved. Thx.

closed 2022-10-24 by bjost2s