OpenRoberta/openroberta-lab #861: multiple robot simulation

Thorlogy, 2020-07-19:
**Is your feature request related to a problem? Please describe.**
 - It would be nice if i could run a single program on multiple robots . 

**Describe the solution you'd like**
 - Choose the program and define / selecet on how many robots this progam should be executed

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - The existing solution is quite is very cumbersome, if you want to change your program, you have to save it each time and then run it as often as you want on as many robots.
- besides, you cannot simply overwrite the program. You have to delete the old program first.

AhwiniMirajkar, 2021-02-03:
Can I work on this

KishkinJ10, 2021-03-26:
hey i am a beginner can i work on this??

RaghuvirShirodkar, 2023-04-28:
Solved

closed 2023-04-28 by RaghuvirShirodkar