OpenRoberta/openroberta-lab #862: Multiple Robot Simulation - Selcted programs

Thorlogy, 2020-07-19:
**Describe the bug**
 - The selected programs are not executed / shown...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'edit / example programs'
2. Click on 'example progams' then click on "multiple robot simulation" 
3. Select programs
4. Click ok
5. Click on the backwards arrow
6. See error

**Expected behavior**
 - The multiple robot simulation should run

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]: Mac
 - OS: [e.g. windows], including version: Mojave 10.14.6
 - Browser [e.g. chrome, firefox], including version: Firefox and Chrome

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2020-11-04:
could not be reproduced, please provide more information or screenshot

bjost2s, 2021-08-12:
seems to be already fixed

closed 2021-08-12 by bjost2s