OpenRoberta/openroberta-lab #886: The lab does not create the correct program link in the text field for some programs

philippmaurer, 2020-08-27:
**Describe the bug**
If you create a program link, the link will be automatically saved in your clipboard. You may choose to manually copy it from a textarea though. If you manually copy it, the link will not work for some programs, yet the automatically created link will work without any problems.
A program with that problem is: Gallery -> Grove_MoistureSensor for Calliope from the user roberta_zentrale

The error message shown fr the manually copied link is: »Grove_MoistureSensor.xml« is not a valid NEPO program and cannot be uploaded!

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Open Roberta Lab and search for "Grove_MoistureSensor" in the gallery
2. Open the Program and create a program link
3. Paste the automatically saved link in another browser tab to see that it works
4. Copy the program link out of the textarea in the program link popup
5. Paste the manually copied link in another browser tab and see the error message.

**Expected behavior**
 - It should not make a difference, whether you manually copy the link or take the automatically copied link. Another solution might be to not show the broken link at all, since it gets copied automatically anways.

**Screenshots**
[image]

**Additional context**
Eventhough the problem seems similar to #65, the metadata of the program link are identical for both program links, so there seems to be a problem in the program itself in this case.

rbudde, 2020-08-27:
2. problem: the URL should start with the server, that generated the link, not always the lab-url. This does not work for dedicated servers
3. problem: microbit-export does not work, see #65 . Replacing µbit by microbit did not solve the problem

philippmaurer, 2020-08-27:
@rbudde Regarding your 3. problem: actually the link in the textarea still spells "µbit" and there is only one ampersand prefixing it. So in order to fix it, you need to replace "µbit" with "&microbit"

rbudde, 2020-08-27:
... but the data copies to the clipboard contains the second & and the text microbit AND does not work ...