OpenRoberta/openroberta-lab #889: missing analog and digital pin configuration of BBC Micro:Bit

nbhsalve82, 2020-09-04:
Dear OpenRoberta Team,

I am learning the first steps of programming and have started yesterday with a BBC Micro:Bit at your homepage lab.open-roberta.org . Yesterday evening the Micro:bit produced a failure and I have seen no mistake in programming code.
Today I have tried the code again and it works. Then I wanted to make a new programming code. Now I am missing the choiceable analog and digital pins! I have to add them manually to the robotic configuration. There are only sensors, buttons and the buzzer ( no analog and digital pins ). 

1. Go to https://lab.open-roberta.org/ and choose MicroBit as robotic
2. Click on Robotic configuration MICROBITbasis
3. See error: missing pin configuration. This has the effect, that I cannot choose a pin under Action/PIN/"Write analog/digital value on PIN "x"
[image]

Expected behavior
I have expected, that in the configuration are listed the pins! But there are no pins configurated!
[image]

**Device information**
 - Type: Desktop PC
 - OS: Windows 10 64-Bit Pro
 - Browser Microsoft Edge and Chrome

boonto, 2020-09-04:
Was the failure yesterday around 21:00 CEST? That's when we created the new release, so that might be the cause.
Do you remember what exactly the error was?

As for the pins, you'll have to add them to the configuration yourself, you can drag them from the toolbox on the left.
We are working on the English documentation for the new configuration, for now here's the [German one with some pictures](https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Konfiguration+Calliope+Mini), which might help.

nbhsalve82, 2020-09-04:
Hi, thanks for response.
The failure has been at approximately 4pm o'clock yesterday.
At 11am o'clock I have tried the code again and it works.
But instead of PIN(0) there has been for example PIN(P0).
Yesterday I had not to configure the interfaces of BBC Micro:Bit.
All interfaces ( Pins, Sensors, Actuator ) has been available without anything.

I am thinking, that it is not well thought out for beginners like primary school students.
The basic configuration with all interfaces is more helpful for the entrance of coding.
If someone want to rename the Pins to more meaningful names, he could change them in the robotic configuartion.
When a beginners starts with questions like "How I can choose the PIN 0/1/2", it is not helpful in my opinion.

Can you change the basic robotic config of BBC Micro:Bit like before update?

boonto, 2020-09-04:
Thanks a lot for the feedback, in our train of thought the pins are more of an advanced feature for more experienced users. That's why they are not in the default configuration.
Let's say you connect an LED to pin 0, you can name it `redLed` and have only the configured actor available in the program, instead of pins 0 to 20 without any context.
With this "configure first, program second" approach the programs can be a lot clearer.

For young students the configuration may be predetermined by the teacher, and the students just have to change the program.

But as you said it might make sense to have the more prominent pins 0, 1 and 2 in the default configuration, we'll discuss this.

bjost2s, 2020-11-04:
We follow the "configure first, program second" approach.

closed 2020-11-04 by bjost2s