OpenRoberta/openroberta-lab #890: New configuration of the calliope. Error with internal sensor/actor, e.g. buzzer

rbudde, 2020-09-07:
since version 4.0.1:

- select calliope mini. The configuration contains the buzzer called "BZ"
- in the program add the block "play frequency". The block refers to "BZ"
- open the simulation. An error pops up "this actuator is not configured"
- nevertheless: the simluation works

second bug:
- in the configuration you can configure a second buzzer "B2" (btw: does this make sense?)
- in the program let the "play" block using this buzzer. Use the dropdown to achiev this.
- open the simulation. The name is changed automatically back to "BZ"

boonto, 2020-09-08:
The first thing is definitely an error and should be fixed.
- [x] Fix wrong error popup

For the second issue we discussed that the internal sensors/actors, that do not have an external equivalent yet:
- [ ] the default blocks should be undeletable and
- [ ] no additional ones can be dragged from the toolbox (either by preventing duplicates or removing them from the toolbox).

This makes more sense and is less confusing for users, if a user cannot add an external buzzer, why should they be able to configure one. They can still rename the default one to add more meaning to a program.

boonto, 2020-09-10:
In addition the buzzer for microbit is now external.

closed 2020-09-29 by bjost2s