OpenRoberta/openroberta-lab #903: Sharing of a program to a group member from a group member behaves unexpected

rbudde, 2020-09-25:
- create a group "g" with some members "01"..."05"
- login as a member "01"
- share a program to another member fails, if you name the user as "02"

- it succeeds, if you name the user "g:02"

Because a group member can only share programs with user of the same group, one solution (among others, maybe better ones) is, that the frontend prefixes the name with "g:"

A second problem is, that the share request is executed **twice**. It should be checked, whether this is a common error pattern and occurs at other places, too.

rbudde, 2020-09-27:
the first problem is solved by a commit I'll push soon.

for the second problem I propose to tell the user to prefix with the group name. This solves the problem for the next weeks.
The frontend logic of sharing must be redesigned. My comment of "share request .... twice" was misleading. **All** sharing data is written to the data base always (you have 8 shares: the frontend issues 8 REST calls). Only the delta should be sent (usually 1 REST call). When this refactoring is done, the group prefix can be added by the front-end.

bjost2s, 2020-09-29:
commit mentioned above is a80563a32626b947f2165d68aae3bf5b2e68e5f3

closed 2020-11-02 by bjost2s