OpenRoberta/openroberta-lab #910: Variable getter and setter have an unwanted dropdown arrow

bjost2s, 2020-10-05:
**Describe the bug**
When creating a single variable (see screenshot), change the name from the default one, take a getter or setter from the toolbox and see the variable name inside. There is a dropdown although only one variable can be chosen.

**Expected behavior**
 No dropdown arrow should be there

**Screenshots**
[image]

 
**Additional context**
 new bug, probably an issue from the last updates.

DhairyaBahl, 2020-10-05:
I would like to work on this project.

rbudde, 2020-10-05:
A second error:
- before you rename a variable, draw a set-variable block out of the toolbox. It has no drop-down. That's ok.
- rename the variable
- add a second variable
- the drop down arrow doesn't appear in the "old" set-variable block. That's wrong.
- but if you click on the variable, the two variables appear. That is unexpected, but allows the name change, at least,.

kirtirajsinh, 2021-03-14:
I would like to work on this issue can you assign it to me.