OpenRoberta/openroberta-lab #923: Calliope Mini Set Pull not working !!!

hzunimd, 2020-10-28:
Problem:

define (in robot configuration) sensor digital 'S' pin 'whatever'
use (in NEPOprog) *2 -- action -- pin -- set pull 'up' on Pin 'none'   -> no Pin selectable!!!!

if i define an output (action: actuator digital) then I can select  'Pin'  but pull(up/down) is for inputs!!

all works fine in old robertalab! (in example 2.7.1)  -> generated c++ code

int main()
{
    _uBit.init();
    
    _uBit.io.P0.setPull(PullDown);
    while ( true ) {
        _uBit.io.P12.setDigitalValue(_uBit.io.P0.getDigitalValue());
        _uBit.sleep(_ITERATION_SLEEP_TIMEOUT);
    }
    release_fiber();
}

bjost2s, 2021-01-21:
The solution could be:

1. Extend the configuration of sensor and actor pins with a dropdown for pullup, pulldown, none. We believe this is a good solution because pullup and pulldown should be defined in the setup once.
2. Mark the pullup, pulldown statement as deprecated, because it doesn't make sence to change this at runtime.

closed 2022-01-05 by bjost2s

RaghuvirShirodkar, 2022-01-06:
Tested okay on `develop`.

Following program was used to test on Calliope mini:
[image]

Robot configuration:
[image]

* Pin P1 was set to pull up.
* Button was connected to pin P1 and ground (-).
* Program was uploaded to Calliope mini.

When the button is pressed, the Calliope shows a heart on it's display, showing that P1 is correctly pulled up to Vcc (+), and that pressing the button grounds it.

Similar test was conducted for pull down, instead checking for digital value on pin P1 = 1 and connecting the button to pin P1 and Vcc (+).