OpenRoberta/openroberta-lab #924: Calliope: "play note" block creates Server Error

philippmaurer, 2020-11-03:
**Describe the bug**
If you use the "play note" block in one of your programs, the server responds with a server error.

**To Reproduce**
Steps to reproduce the behavior:
1. Celect Calliope as your robot
2. Drag the "note play whole note c" block into your program
3. Press the play button or start the simulation
4. See error

**Expected behavior**
A program should be created, that plays the specified note

**Screenshots**
[image]

**Device information**
 - Server 4.0.5

**Additional context**
Instead of "play note" the block reads "note play". Is this a wanted behavior?

bjost2s, 2020-11-03:
```
de.fhg.iais.roberta.util.dbc.DbcException: Could not invoke method jaxbToAst for block mbedActions_play_note with fields ACTORPORT _B 
	at de.fhg.iais.roberta.transformer.Jaxb2ProgramAst.invokeMethod(Jaxb2ProgramAst.java:98)
	at de.fhg.iais.roberta.transformer.Jaxb2ProgramAst.invokeJaxbToAstTransform(Jaxb2ProgramAst.java:72)
```

Gillanius, 2020-11-03:
Same behaviour for "get angle gyroscope"

[image]

Belongs this error message to the same issue?

closed 2021-01-08 by bjost2s