OpenRoberta/openroberta-lab #927: I2C support for Calliope mini

MKleinSB, 2020-11-15:
**Is your feature request related to a problem? Please describe.**

As a necessary feature for adding hardware by myself  via I2C it would be nice to have I2C support in OpenRoberta.

And it would be nice to add the Hardware/Blocks/Function in the Configuration as an actor or sensor. Perhaps via 
load Sensor/Action.

Thanks

bjost2s, 2020-11-16:
Thank you for this issue, we do already support some sensors / actuators with I2C, but for a generic support I think we also have to introduce byte as a datatype and bit operations. Can you give us a example how you would like to use these blocks?

rbudde, 2020-11-16:
my 2 cents: we discussed this topic several times. We would like to support generic I2C, not only for the Calliope, for the arduinos the need is as urgent as well. @BeateJ mentioned one consequence for the lab:
- support of byte and bit ops. As many beginners are working with the lab, we would need to support many toolboxes (to avoid flooding beginners with these blocks)
- furthermore I2C sensors/actors come with a (small) libraries. With this we could avoid the low-level stuff, but then we have to think about a way to extend the lab (in a safe, light-weight manner) with such libraries, maybe with a possibility to add self-defined blocks (this would make the use of new devices much more easy)

Currently we don't have the time to work on this. If you describe a use-case and a technical solution very detailed, we could try to find somebody (a student?) working on this (which is the way to contribute to OSS anyway :-) )

MKleinSB, 2020-11-21:
@rbudde hat Recht. Neben den I2C Funktionen zum Lesen und Schreiben werden die verschiedenen Datentypen und Bitverknüpfungen benötigt.  Genauso wichtig ist ein System um für den User Blöcke zu definieren, sodass dieser gar nicht selbst den Sensor über I2C programmieren muss. So wie dies bei MakeCode über die Möglichkeit der Erweiterungen in github möglich ist.
Beim Einsatz mit Schulklassen kommt es darauf an, dass die Sensoren einfach anzusprechen sind. Wenn man sich z.B. den Farbsensor anschaut hat man zwar viele Einstellungsmöglichkeiten, für den Gebrauch an Schulen wird aber eigentlich nur eine einzige benötigt.  Wenn man dann den Farbsensor in Klasse 3-7 verwenden möchte stellt man fest, dass Probleme auftreten, die man auf dem Niveau der Schüler:Innen nicht lösen kann. Z.B. liefert der Farbsensor je nach Batteriestand, Umgebungslicht, Papier oder Farbauftrag andere Werte, sodass man mit einprogrammierten Farbwerten Farben oft nicht wiedererkennen kann. Das ist für die Schüler:Innen sehr frustrierend. Man benötigt also eine gewisse Toleranz und eventuell einen Farblernmodus um Farben abzuspeichern. Hier wären dann Funktionen, die man vielleicht in einem 3. Menü, neben Beginner und Expert einladen könnte, praktisch. Lösungen kann ich leider auch nicht anbieten ☹
Gerade wegen der Möglichkeiten Erweiterungen einzuladen, verwenden Lehrer die mit ihrer Klasse mehr als die Grundfunktionen des Calliope mini verwenden möchten MakeCode.

rbudde, 2022-05-11:
due to timing restrictions, currently impossible to implement

closed 2022-05-11 by rbudde