OpenRoberta/openroberta-lab #933: New configuration seems to change positions of the board's svg

bjost2s, 2020-12-09:
here is the standartconfiguration of calliope before storing
[image]
which is ok 
and here how it looks after storing and loading again
[image]

aykborstelmann, 2020-12-14:
I found more problems with this way of the configuration svg block:

e.g. 
[image]

which is not that big of a problem, since it isn't saved anyway.

Or 

[image]

which is a big problem, since it causes a server error.

aykborstelmann, 2020-12-30:
The problem should be solved. For testing purposes I have created some examples (exported and stored in the db).
[issue-933-testing.zip](https://github.com/OpenRoberta/openroberta-lab/files/5753942/issue-933-testing.zip)

In the DB is one user, username `Ayk` and password `AykAyk` who has programs for Calliope, Microbit and Arduino.

commit 2179fc0a (2020-12-30): Fix #933 for Calliope

Modify default configuration to contain robot xml block and add block type to yml config
commit 62af5aff (2020-12-30): Fix #933 for Microbit

Modify default configuration to contain robot xml block and add block type to yml config
commit 63def8ba (2020-12-30): Fix #933 for Arduino

Modify default configuration to contain robot xml block and add block type to yml config
Modify Arduino Visitors to ignore new robot xml block
commit 80a8d35c (2020-12-30): Fix #933 for Arduino

Modify default configuration to contain robot xml block and add block type to yml config
Modify Arduino Visitors to ignore new robot xml block
closed 2021-01-22 by bjost2s