OpenRoberta/openroberta-lab #949: micro:bit V2 support

microbit-carlos, 2020-12-23:
Hi all,

Wasn't sure the best way to share this info, so I thought an issue here might be a good way.
The micro:bit V2 has is already available and we've updated the JS libraries to help porting editors to the new board.

If Open Roberta is already using microbitFs, then all you'd have to do is update to the latest version of the library and update a couple of calls (the MicropythonFsHex constructor to now also add a V2 hex, and there is a method to generate a "Universal hex" file).
- https://github.com/microbit-foundation/microbit-fs
- https://microbit-foundation.github.io/microbit-fs/quick-guide.html

An example can be found in the Python Editor:
- https://github.com/bbcmicrobit/PythonEditor/blob/16e2ffff11958488f43a957b0a602db5b49cc1fe/js/fs.js#L68
- https://github.com/bbcmicrobit/PythonEditor/blob/16e2ffff11958488f43a957b0a602db5b49cc1fe/js/python-main.js#L931

The latest version of the hex files for V1 and V2 can be found in their respective repos:
- V1: https://github.com/microbit-foundation/micropython-microbit-v2/releases/
- V2: https://github.com/microbit-foundation/micropython-microbit-v2/releases/
    - We plan is to push the V2 port to the upstream MicroPython repository when it's ready, so this is a temporary repository until everything goes upstream

The V2 uses a slightly different microcontroller (nRF52 with a Cortex M4 instead of the V1 M0+), so we've created a new hex file format called Universal Hex. This hex format is capable to store the data for both microcontroller, so we can generate one file that works everywhere.
The microbitFs library already generates Universal Hexes, but if you interested there is also a new library specifically to combine two Intel Hex files into a Universal Hex:
- Library: https://github.com/microbit-foundation/microbit-universal-hex/
- More info about the Universal Hex format: https://tech.microbit.org/software/hex-format/

Let me know if you have any questions.

Happy holidays!

rric, 2023-01-11:
+1

The longer this issue is open, the more urgent it gets. Is there yet a plan to resolve it?

heini208, 2023-03-01:
From now on this will be discussed in Ticket: #1461

closed 2023-03-28 by bjost2s