OpenRoberta/openroberta-lab #960: Redesign of the buttons in the simulation tab

rbudde, 2021-01-28:
to be defined.

Possibilities:
- move debug button next to run
- if debug enabled, put debug buttons between run + start debug
- remove the run to next breakpoint (use the run button for that)

suraj10k, 2021-01-28:
hey can i work on this?

rbudde, 2021-01-29:
you are welcome to do that. Some remarks:
- it is essential, that you first play with the simulator and debugger with simple programs of 5-10 blocks to become familiar with both
- then first design the new button layout and discuss this with us
- the button images are from the typicon font. It is possible to use one of the existing, but it's possible to generate new from a SVG. You can do this (requires probably search in the net), but we can help.

suraj10k, 2021-02-06:
can i open a pr?

suraj10k, 2021-02-07:
Hey can you pls guide me to the file that contains the code for the simulator buttons

bjost2s, 2021-02-08:
@suraj10k , have you already a design in your mind? Please share it before starting implementation

suraj10k, 2021-02-22:
i was thinking we could center the buttons (i felt they were  a little too close to the buttons from the robot configuration menu) also as you rightly said i too feel it might be better to move the debugger button next to the play button. one more thing i thought might be good is if we move the change environment button to be located inside or somewhere close to the environment that gets changed.

suraj10k, 2021-02-22:
also i tried looking up icons for the buttons , but i found it hard to find a batch of icons for all the existing buttons , that had a uniform theme(look).
i could look into it more if your interested in changing the icons