OpenRoberta/openroberta-lab #966: Calliope list of strings comparison compiler error

bjost2s, 2021-02-18:
The following program cannot be compiled:
[image]

rbudde, 2021-02-18:
the complete error message :-). Note, that other types (bool, number) work fine. Strings (instead of list<string>) work, too. The ++ operator is defined for ManagedString

```
1/3] Building the NEPOprog. In file included from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattClient.h:22, from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/BLE.h:23, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/bluetooth/MicroBitBLEManager.h:42, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:69, from 
/tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/ServiceDiscovery.h: In member function 'virtual ble_error_t 
ServiceDiscovery::reset()': /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/ServiceDiscovery.h:148:77: warning: implicitly-declared 'UUID& UUID::operator=(const UUID&)' is deprecated 
[-Wdeprecated-copy] 148 | matchingServiceUUID = UUID::ShortUUIDBytes_t(BLE_UUID_UNKNOWN); | ^ In file included from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattService.h:20, 
from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattServer.h:21, from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/BLE.h:22, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/bluetooth/MicroBitBLEManager.h:42, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:69, from 
/tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/UUID.h:164:5: note: because 'UUID' has user-provided 'UUID::UUID(const 
UUID&)' 164 | UUID(const UUID &source) { | ^~~~ In file included from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattClient.h:22, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/BLE.h:23, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/bluetooth/MicroBitBLEManager.h:42, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:69, from /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: 
/opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/ServiceDiscovery.h:150:77: warning: implicitly-declared 'UUID& UUID::operator=(const UUID&)' is deprecated [-Wdeprecated-copy] 150 | 
matchingCharacteristicUUID = UUID::ShortUUIDBytes_t(BLE_UUID_UNKNOWN); | ^ In file included from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattService.h:20, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/GattServer.h:21, from /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/BLE.h:22, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/bluetooth/MicroBitBLEManager.h:42, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:69, from 
/tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: /opt/ora-cc-rsc/RobotMbed/libs2017//ble/ble/UUID.h:164:5: note: because 'UUID' has user-provided 'UUID::UUID(const 
UUID&)' 164 | UUID(const UUID &source) { | ^~~~ In file included from /opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/list:63, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/NEPODefs.h:70, from /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:4: 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/bits/stl_list.h: In instantiation of 'bool std::operator==(const std::__cxx11::list<_Tp, _Alloc>&, 
const std::__cxx11::list<_Tp, _Alloc>&) [with _Tp = ManagedString; _Alloc = std::allocator]': /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:20:19: required from 
here /opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/bits/stl_list.h:2004:56: error: no match for 'operator==' (operand types are 'const 
ManagedString' and 'const ManagedString') 2004 | while (__i1 != __end1 && __i2 != __end2 && *__i1 == *__i2) | ~~~~~~^~~~~~~~ In file included from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:43, from /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/types/ManagedString.h:264:10: note: candidate: 'bool ManagedString::operator==(const ManagedString&)' 264 | bool operator== 
(const ManagedString& s); | ^~~~~~~~ /opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/types/ManagedString.h:264:10: note: passing 'const ManagedString*' as 'this' argument 
discards qualifiers In file included from /opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/utility:70, from 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/array:38, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit-dal/inc/core/MicroBitCompat.h:36, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/MicroBit.h:40, from /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:3: 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/bits/stl_pair.h:448:5: note: candidate: 'template constexpr bool std::operator==(const 
std::pair<_T1, _T2>&, const std::pair<_T1, _T2>&)' 448 | operator==(const pair<_T1, _T2>& __x, const pair<_T1, _T2>& __y) | ^~~~~~~~ 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/bits/stl_pair.h:448:5: note: template argument deduction/substitution failed: In file included from 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/list:63, from /opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/NEPODefs.h:70, from 
[... cut ...]
template argument deduction/substitution failed: In file included from /opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/list:63, from 
/opt/ora-cc-rsc/RobotMbed/libs2017//microbit/inc/NEPODefs.h:70, from /tmp/openrobertaTmp/-Ao08-WSr-o1/NEPOprog/source/NEPOprog.cpp:4: 
/opt/compiler/gcc-arm-none-eabi-9-2019-q4-major/arm-none-eabi/include/c++/9.2.1/bits/stl_list.h:2004:56: note: 'const ManagedString' is not derived from 'const SafeBool' 2004 | 
while (__i1 != __end1 && __i2 != __end2 && *__i1 == *__i2) | ~~~~~~^~~~~~~~ Error in building !!!
```

PamoCurtis, 2023-03-07:
Tested on dev server:
- Calliope Mini: Successful
- Calliope Mini Blue: Successful
- Calliope Mini 2016: **Unsuccessful** 

Program: [Calliope_Issue966.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/10908846/Calliope_Issue966.xml.txt)
Error Message: [Error_Msg_Calliope_Mini_2016_Issue966.txt](https://github.com/OpenRoberta/openroberta-lab/files/10908860/Error_Msg_Calliope_Mini_2016_Issue966.txt)

RaghuvirShirodkar, 2023-03-07:
Ah it seems I missed updating the NEPODefs header file for Calliope Mini 2016 version. A [pull request](https://github.com/OpenRoberta/ora-cc-rsc/pull/13) has been raised for the same in `ora-cc-rsc` repository.

PamoCurtis, 2023-03-07:
Tested the [pull request](https://github.com/OpenRoberta/ora-cc-rsc/pull/13) locally and works as intended now!

RaghuvirShirodkar, 2023-03-08:
Tested successfully on test server with Calliope Mini 2016.

closed 2023-03-08 by bjost2s