OpenRoberta/openroberta-lab #967: Gallery of Background Images. Easy Access of Images

rbudde, 2021-02-19:
we would like to have a wiki page, which describes in detail the requirements for background images (standard sizes, relation to robot size, ...).

Then there should be a gallery of images, which can be used for free. There have been proposals in the context of GCin, which can be used as a starting point.

The load of a background image in the lab should be possible from a user supplied URL. This makes the use of the gallery easy. Furthermore a user, a teacher e.g., could host images somewhere, a learning platform e.g., which may be used in exercises by other user, students e.g..

Examples of pictures in the gallery should include:

- small and large areas
- color patches for programs as "count the number of colored areas"
- rooms for programs as "construct a map" (is this feasible?)
- areas for multiple robot application (for instance: first robot follows a path and activates a second one using the bumper, which in turn follows another path)

Please add more examples in comments

mayank0508, 2021-02-21:
hey can I work on this issue ??

johnmusk, 2021-02-22:
i wanna work on this

bjost2s, 2021-03-02:
@mayank0508 and @johnmusk please contact us in gitter https://gitter.im/openroberta-lab/GSoC

mayank0508, 2021-03-02:
👍🏽

ensonic, 2021-03-04:
Each image should have a 'copy link' (chain icon) button to make it easy for people to get the url.