OpenRoberta/openroberta-lab #969: named configuration loaded correctly with a program, but the name is then forgotten

rbudde, 2021-02-25:
- create a named configuration
- create a program using it
- store the program (implicitly together with the named configuration)
- change the named configuration
- load the program again. The change is visible, i.e. the named configuration is loaded from the database

- but: the name of the configuration is NOT shown in the configuration tab
- if the configuration is then changed, only "store as" is enabled. The name of the configuration seems to be lost.
- and: if the program is stored again and then reloaded, the configuration is unnamed, i.e. the configuration name has not been stored with the program, only the configuration itself.

EthicalCoder99, 2021-02-26:
I am exploring the bug. If we create a named configuration and save it along with the program, next time when we load the program the Configuration name is not visible on the configuration tab. If we change any configuration we can't save it. But if we open the Configuration that we have created along with the program, i.e by going to Edit -> My Configuration -> Your_Configuration_Name. Then if we edit the configuration we can save those configurations now.
I want to work on this bug. Can you give me some entry points to track this bug?

rbudde, 2021-03-03:
your pull request solved the problem described in the issue, during testing we found, that your changes will put the name of the default configuration into the tab always. This is not correct for the following reason:

1. case (works with your changes) the user uses a named config. The frontend sends only the config name to the server, the server gets the config from the db and sends back BOTH the config and the config name. The config name must be shown in the tab
2. case (works with your changes) the user uses the default config (shown name: ROBOTNAMEbasis). The frontend sends no config and no config name to the server [this behavior is wrong, see 3: IT SHOULD SEND INFO TO THE SERVER, THAT SHOWS THAT THE DEFAULT IS USED, e.g. the default config name]. This is a frontend change. The server must sent back the default config name and the config. For this the server must detect, that a default config name is used (e.g. by checking the name). This is a server change. The default config name must be shown in the tab
3. case (fails with your change) the user uses a MODIFIED default config (NO name is shown). The frontend should send the config and NO config name to the server. The server should sent no name and the config back. No name must be shown on the tab.

The pull request will show the default name in both caes 2 and 3. The required changes described above. Server changes there are not that easy. Please drop a note, whether you want to work on that or not. In the latter case I'll complete your pull request with the needed server changes.

Thank you very much for the pull request!

EthicalCoder99, 2021-03-03:
I will try to make the server changes. Can you confirm that whether hot reloading is supported or not. Because every time I make some changes in .js files I have to clear the site cache to reflect those changes.

EthicalCoder99, 2021-03-03:
Ok I will look into it. Can you assign me this issue.

bjost2s, 2021-03-04:
@EthicalCoder99 indeed you have to clear the cache each time, but you can configure this somewhere e.g. in the chrome debugger