OpenRoberta/openroberta-lab #995: EV3 Sim: Robot turning with fine degrees not reproducible when using it in while loop

germanicianus, 2021-04-06:
When using the while loop shown on below screenshot, the simulated robot turns to a different direction in every simulation run. Furthermore, the calculation error sums up which is kind of expected.
[image]

germanicianus, 2021-04-06:
This issue is already reproducible with a simple single turn of 90 degrees. In most cases the simulated robot slightly over-turns - see screenshot below.
[image]

rbudde, 2021-04-08:
the accuracy of turns is a problem for both simulation (frame generation at discrete steps) and real robots (friction). There is a text in our FAQ https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Robotersysteme describing the problem and how to solve it: use sensor data (which is available for both simulation and real robot) for a correction step at low speed.

germanicianus, 2021-04-10:
@rbudde, thanks for the explanation - I missed that part in the documentation.

closed 2021-04-10 by germanicianus