OpenRoberta/openroberta-lab #997: UI Change: Start / Stop Buttons in the Debugging Tab

rbudde, 2021-04-08:
I think, the new design of the start/stop buttons for the debugger is superior to the old one, but may generate irritations when using.  The start button is the same as the button "go to next breakpoint". That is fine.

The red stop button is enabled after start and can be used to stop the running program. That is fine, too.

A program running on the **Calliope** in debugging mode does not terminate. Thus, before one can restart the program for a second run, the red stop button *must* be pressed first, then the start button can be used. If you forget this, you may believe, that the simulation is frozen.

**Proposal:** if the last block of a program is executed, this should be remembered as "readyToBeRestarted". The program may "continue" to run as before. Now, if the start button is pressed and "readyToBeRestarted" is true, internally the sequence stop (as if the red button was pressed) followed by (re-)start should be triggered. This should be not too complex to implement.

LawfulTheBest, 2021-05-05:
Sir @rbudde could you please guide me on how to start working on this issue?

heini208, 2021-05-26:
A few Questions about the desired looks of the change:
Should the "next-Breakpoint" button become Red or stay Blue?
Should the other Stop button disappear?
Tell me which version looks best:
V1:
[image]
V2:
[image]
V3:
[image]

rbudde, 2021-05-26:
I'd prefer V1

closed 2021-08-12 by bjost2s