OpenRoberta/openroberta-lab #1025: export all programs of a user as a zip file

rbudde, 2021-05-25:
a user may want to have a backup of his/her programs, which he/she has stored in the data base.

Workflow:
- login
- select menu entry "export all programs as zip file"
- a file selection dialog appears. The user selects a fiel name
- the programs are stored

Remarks:
- the place in the menu should be discussed
- other formats (tar, ...) could be an option

heini208, 2021-05-26:
Shouldn't there also be an "import zip to my programs"? Because if not, Loading a back up is just as tedious as currently creating one.

rbudde, 2021-05-26:
yes, it should be symmetric. But, importing has the risk of an DOS attack by importing millions of programs. Maybe we need a limitation of the number of programs (100, 1000?)

heini208, 2021-05-27:
should export also export all Programs that are shared with the user including the programs of group members of those groups, that are owned by the user? like in the getProgramInfoOfProgramsOwnedByOrSharedWithUser() from [ProgramProcessor.java](https://github.com/OpenRoberta/openroberta-lab/blob/c2060de82a50d780cd656519b3e7845e30722f0b/OpenRobertaServer/src/main/java/de/fhg/iais/roberta/persistence/ProgramProcessor.java#L242) (but instead of for one Robot for every Robot or should it only include the Programs owned by the User

rbudde, 2021-05-27:
I think, we should discuss this. I propose:
- to save NOT the programs that are shared by other users (because they are responsible to backup them),
- but, to save the programs of all members of all groups the user owns

The problem is to see this. The only possibility I can imagine is to encode the information into the name of the member, e.g.
- my programs: <original name>
- group member programs: <group>_<member name>_<original name>

Ideas?

heini208, 2021-05-28:
Something along those lines is already being done [here](https://github.com/OpenRoberta/openroberta-lab/blob/c2060de82a50d780cd656519b3e7845e30722f0b/OpenRobertaServer/src/main/java/de/fhg/iais/roberta/persistence/ProgramProcessor.java#L372) if  I'm not mistaken just instead of loadall(user,robot) it has to be loadAll(user). I think its best to put those parts into their own helper methods and use them maybe overload them with (user) and (user robot). The method this is in is way too long anyways.

heini208, 2021-06-23:
Due to the nature of how this feature is being implemented right now, it is possible to check if a program is the default program with no changes to the config and a check for default ProgramText shouldn't be too difficult either , i suggest to not Download programs that are unchanged, so you don't download unnecessary files. With this you could also use the feature to clean up your workspace.
But i can also see this causing confusion if you don't know why certain Programs are missing, and if you have a certain naming structures (like numbering your programs) this structure would have holes when you had a few empty programs within it.

heini208, 2021-06-29:
I have thought about the possible Folder Structure please let me now what combination of general folder structure and group structure is prefered, i like: general: V2 with group: V3
General Structure V1:
[Export_structureV1.pdf](https://github.com/OpenRoberta/openroberta-lab/files/6733323/Export_structureV1.pdf)
General Structure V2:
[Export_structureV2.pdf](https://github.com/OpenRoberta/openroberta-lab/files/6733326/Export_structureV2.pdf)

Group Structure:
[GroupStructures.pdf](https://github.com/OpenRoberta/openroberta-lab/files/6733330/GroupStructure.pdf)