OpenRoberta/openroberta-lab #1026: add a REST method to do xml transformation from old to actual xml format

rbudde, 2021-05-25:
**Context**:
- either in the data base or exported there may be outdated xml programs
- if this is detected, a transformer is called to update the xml to the current version 3.1
- currently this is done inline in rest methods that are responsible for loading/importing program

**Goal**:
- create a new rest method, that gets old xml and returns new xml
- add the uri of this rest method to the openroberta.properties
- the  rest methods, which detect the need of tranformation, should call the new rest method

**Reason**:
- the old method is not flexible, because the transformer makes it hard to update our AST classes
- this issue is a prerequisite for moving the transformer to a docker container of its own

afif1400, 2021-05-26:
Can I work on this

rbudde, 2021-05-27:
@afif1400 yes, of course, you are very welcome. I hope, the work is not too hard, but probably the setup of the workspace, etc. may be time-consuming. Some remarks:

- you should follow our README to setup the workspace. You must NOT install cross compilers (c++, ...), as you must not compile generated programs for a robot.
- the preferred way for questions is to use gitter: https://gitter.im/openroberta-lab/community, but mail is fine, too.
- I prefer to discuss a short sketch about proposed changes before implementation to avoid oversights, misundertanding, etc. But this varies depending on experience.
- your work should be submitted as pull request against develop.

Enjoy :-)

bjost2s, 2024-10-24:
no time to work on

closed 2024-10-24 by bjost2s