OpenRoberta/openroberta-lab #1033: Simulation accuracy depends on speed of movements

aykborstelmann, 2021-05-31:
As you can see while the simulation is correct when using the speed `2` the simulation is in correct (rotates to 180°) with speed `90` it rotates to 186°. The accuracy is antiproportional to the speed of the movements.

Our simulation should be correct independent of the speed. 

**Low speed**
[image]
[image]

**High speed**
[image]
[image]

rbudde, 2021-06-01:
Have a look at our FAQ and open the first entry of the page: https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Robotersysteme

aykborstelmann, 2021-06-10:
Okay, I didnt know that

closed 2021-06-10 by aykborstelmann