OpenRoberta/openroberta-lab #1041: "For each" with colors does not work as expected, simulation only

bjost2s, 2021-06-10:
The following program can be executed, but the behaviour of the robot is unexpected:
[image]
The robot starts driving backwards (starting on the white background) and the program execution terminates when the robot reaches the black line.

The values of the list elements seems to be correct, probably a problem of the "for each" block in the simulation execution.

Note: this program was working fine in the beginning of this year.

rbudde, 2021-06-10:
error messages from console:
```
interpreter.util.js:14 invalid expr op: ledColor
interpreter.util.js:15 Uncaught invalid expr op: ledColor
interpreter.util.js:14 pop failed with empty stack
interpreter.util.js:15 Uncaught pop failed with empty stack
interpreter.util.js:14 setVar failed for: f
interpreter.util.js:15 Uncaught setVar failed for: f
interpreter.util.js:14 getVar failed for: f
simulation.js:487 END of Sim
```
probably an error introduced when the code generation of loops has been stratified

rbudde, 2022-05-04:
works as expected

closed 2022-05-10 by bjost2s