OpenRoberta/openroberta-lab #1042: Inconsistent Rotation Values

SpartanTeam, 2021-06-10:
**Describe the bug**
Turn blocks do not make the robot rotate the expected value. When attempting to rotate 90° from the starting point with a rotation of 0°, the robot will stop at a 97°, 92° or sometimes even numbers below 90°, instead of exactly 90° as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Action
2. Add a "Turn (right) speed (50) % degree (90°)"
3. Open simulation view
4. Open sensor data view
5. Start the simulation
6. Take note of the robot's degree (which will hardly be 90°)
7. Reset the simulation
8. Repeat steps 5 to 7 some times and notice how the same code will produce different rotation values.

**Expected behavior**
Turn blocks should rotate exactly 90 degrees every time the program is executed.

**Screenshots**
[image]

**Device information**
 - Type: Desktop (Also observed on Mobile)
 - OS: Windows 7 Ultimate 6.1 (Also observed on Windows 8, 10 and Android)
 - Browser: Chrome 90 on Windows 7

**Additional context**
While creating programs to participate in tournaments, we have observed this issue regarding the rotation. If it's not a bug, and is, in fact, a mistake in our part, please point us towards how to fix this issue as the help section doesn't clarify anything about rotation being imprecise. Thanks.

Ch40sl0rd, 2021-06-11:
You can find an explanation as to why this happens here: [https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Robotersysteme?showLanguage=en_GB].

tldr: the higher the rotation speed, the higher the robot turns in one frame. since we cannot check half frames, we suggest lowering your rotation speed.

rbudde, 2021-06-11:
thank you for the issue and the first answer. The essence of the link is: real robots and sim robots cannot achieve exact turns without sensor data. Thus: rotate fast, compute the error, correct slow. This is an inherent problem of embedded/robot systems: you have sensor->compute->actor loops and this takes time and the environment evolves independant from the system :-)